QIDITECH/QIDI_PLUS3 #3: Code for new releases not available

stew675, 2023-10-11:
I noticed that for the last few releases that the source code base hasn't been updated other than updates to the README.md file

Is it possible to have the source code changes for the latest releases be pushed to the repositories?

Rainboooom, 2023-10-14:
[image]
Because we want everyone to have a direct view of the upgrade files and necessary processes you need at first glance, I synchronized the latest version of the main branch's read.me file. If you need to see the latest version of the source code, just switch to the corresponding branch in the upper left corner.

stew675, 2023-10-14:
Ah, thank you for that.  I'm generally familiar with the default branch also being synced with the latest release branches.

closed 2023-10-14 by stew675