QIDITECH/QIDI_PLUS3 #6: Ther is no Downlod Image for V 4.2.12

lorangthierry, 2023-11-29:
Hi,

"4.2.12 Update content" in the README and on the <code> site
but under releases the latest Version is Plus3_V4.2.10

Regards
Thierry

noisymime, 2023-12-04:
Looks like the code has been removed from the v4.2.12 branch as well. 

Hoping the 4.2.12 version is OK as I downloaded and installed it last week when it was available.

QIDITECH, 2023-12-04:
We discovered some minor issues after the release of V4.2.12, which have now been fixed.

doomcrewinc, 2023-12-05:
@lorangthierry I have the white display issue, how did you resolve so I can fix my printer

QIDITECH, 2023-12-05:
@doomcrewinc Please contact our after-sales service and they will give you detailed tutorials.
https://qidi3d.com/pages/warranty-policy-after-sales-support

doomcrewinc, 2023-12-05:
@QIDITECH I did I'm waiting on an email responose back