QIDITECH/QIDI_PLUS3 #9: firmware update problem

vojd11, 2023-12-28:
You forget to add QD_factory_mode.txt file inside the rar archive. Without it, qidi x-plus 3 updates were not starting.

QIDITECH, 2023-12-30:
This is the file used to restore factory Settings, firmware updates do not need it. Your inability to update should be caused by other reasons, such as the update file path error, file corruption, etc. Please provide more detailed information.

jjl40, 2024-01-02:
I had a problem with installing 4.2.12. Would not start when thumb drive was inserted. The bottom menu button on the touch screen I noticed an "update" button that started the new install.

gptech2444, 2024-01-30:
This latest FW update doesn't work or your instructions are wrong.
Extracting all the files or the folder to the root of the USB makes no difference. FW update doesn't happen and there's no option action an update.

CChen616, 2024-01-30:
The latest update package in the new release has been updated.  We are working on getting the update notes polished. Sorry for any inconvenience this might have caused.

closed 2024-01-30 by CChen616