QIDITECH/QIDI_PLUS3 #15: Fans in fluidd interface

next-comp, 2024-01-30:
Can you please rename in fluidd interface:
Fan 0
Fan 2
Fan 3
to something meaningful?

_My understanding is that:
Fan 0 - part cooling fan
Fan 2 - Rapid Cooling / Auxiliary fan (in chamber, on the side)
Fan 3 - Chamber exhaust fan (with activated carbon filtering)_

CChen616, 2024-01-31:
We plan to fix this issue in the next few versions. Thank you for your suggestion!

leadustin, 2024-07-15:
> We plan to fix this issue in the next few versions. Thank you for your suggestion!

Any update progress?

CChen616, 2024-07-16:
Due to the numerous suggestions we've received regarding the security of the printer system, we are currently making adjustments to the relevant updates. Additionally, there have been changes in the UI team, so it may take some more time.

leadustin, 2024-08-19:
Can someone please explain to me exactly which fans these are? Hotend fan is clear to me. the other two are the mini fans on the left and back of the print head?

[heater_fan hotend_fan]
pin:MKS_THR:gpio1
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

[heater_fan hotend_fan2]
pin:MKS_THR:gpio20
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

[heater_fan borad_fan]
pin:PC4
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

leadustin, 2024-08-20:
> Can you please rename in fluidd interface: Fan 0 Fan 2 Fan 3 to something meaningful?
> 
> _My understanding is that: Fan 0 - part cooling fan Fan 2 - Rapid Cooling / Auxiliary fan (in chamber, on the side) Fan 3 - Chamber exhaust fan (with activated carbon filtering)_

It is possible that it does not work with the old version of Klipper. It definitely works with Klipper 0.12. Make a backup of the printer.cfg and then open the file. Search for this section

##主板上大涡轮风扇
[output_pin fan2]
pin: PA8
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0.00
scale: 255
shutdown_value: 0.0

REPLACE WITH:

[fan_generic auxiliary_cooling_fan]
pin: PA8
shutdown_speed: 0.0
cycle_time: 0.100
hardware_pwm: false
kick_start_time: 0.100
off_below: 0.0

SEARCH FOR THIS:

##活性炭吹风扇
[output_pin fan3]
pin:PC9
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0
scale: 255
shutdown_value: 0.0

REPLACE WITH:

[fan_generic chamber_circulation_fan]
pin:PC9
shutdown_speed: 0.0
cycle_time: 0.100
hardware_pwm: false
kick_start_time: 0.100
off_below: 0.0

SEARCH FOR THIS:

## FAN1  吹模型风扇
[output_pin fan0]
pin: MKS_THR:gpio2
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0
scale: 255
shutdown_value: 0.0

REPLACE WITH:

[fan_generic cooling_fan]
pin: MKS_THR:gpio2
shutdown_speed: 0.0
cycle_time: 0.100
hardware_pwm: false
kick_start_time: 0.100
off_below: 0.0

SEARCH FOR THIS:

[gcode_macro M106]
gcode:
    {% if params.P is defined %}
      {% if params.S is defined %}
        SET_PIN PIN=fan{params.P|int} VALUE={params.S|int}
      {% else %}
        SET_PIN PIN=fan{params.P|int} VALUE=255
      {% endif %}
    {% endif %} 

    {% if params.T is defined %}
        {% if (params.T|int)==-2 %}
          {% if params.S is defined %}       
            SET_PIN PIN=fan3 VALUE={params.S|int}
          {% else %}
            SET_PIN PIN=fan3 VALUE=255
          {% endif %}
        {% endif %}
    {% endif %}

    {% if params.P is undefined %}
    {% if params.T is undefined %}
    {% if params.S is defined %}
            SET_PIN PIN=fan0 VALUE={params.S|int}
       {% else %}
            SET_PIN PIN=fan0 VALUE=255       
    {% endif %}
    {% endif %}
    {% endif %}

[gcode_macro M107]
gcode:  
    SET_PIN PIN=fan0 VALUE=0 
#    SET_PIN PIN=fan1 VALUE=0
#    SET_PIN PIN=fan2 VALUE=0

REPLACE WITH THIS:

[gcode_macro M106]
gcode:
    {% set p = params.P|default(0)|int %}

    {% if p == 2 %}
    {% if params.S is defined %}
        SET_FAN_SPEED FAN=auxiliary_cooling_fan SPEED={(params.S|float / 255.0)}
    {% else %}
        SET_FAN_SPEED FAN=auxiliary_cooling_fan SPEED=1
    {% endif %}
    {% endif %} 

    {% if p == 0 %}
								   
    {% if params.S is defined %}
        SET_FAN_SPEED FAN=cooling_fan SPEED={(params.S|float / 255.0)}
    {% else %}
        SET_FAN_SPEED FAN=cooling_fan SPEED=1
					 
				   
    {% endif %}
    {% endif %} 

    {% if p == 3 %}
								  
    {% if params.S is defined %}
        SET_FAN_SPEED FAN=chamber_circulation_fan SPEED={(params.S|float / 255.0)}
    {% else %}
        SET_FAN_SPEED FAN=chamber_circulation_fan SPEED=1
    {% endif %}
    {% endif %}

[gcode_macro M107]
gcode:  
    SET_FAN_SPEED FAN=cooling_fan SPEED=0

CChen616, 2024-08-21:
These changes are based on version 0.10.0 of Klipper. The main issue is that the control system for the 3 Series machines, Xindi, uses hard-coded names for the fans. This means that if you only modify the `printer.cfg`, the fans will still function, but they won't be controllable via the screen (though this won't affect Fluidd).

The specific mapping is just like the one you provided.

For now, without modifying Xindi, the easiest solution is to use the command `M106 P{number}` to test which fan corresponds to which name and make a note of it.

Phil1988, 2025-01-24:
@next-comp 
They are named sidefan and filterfan on [**FreeDi**](https://github.com/Phil1988/FreeDi).
May this be something for you?