QIDITECH/QIDI_PLUS3 #16: Z-Offset not saving

defiante, 2024-01-31:
Several users reporting that after the latest update the Z-Offset does not get saved after running an auto bed level or Set Z-Offset from the display.

https://www.reddit.com/r/QIDI/comments/1absv5a/v4213_firmware_problems/

CChen616, 2024-02-01:
Try to do a factory reset. If this does not resolve the issue, please provide more details about the circumstances and relevant information when the problem occurs.

WMU-JBLAINE, 2024-02-08:
I can confirm that factory reset does not fix this issue.  This happens every time after setting the z offset. This is tested on X-Plus 3, I'll do a z-off set calibration then go to print and the head is starting higher then what I calibrated for.  Go back to calibration and i'm having to set the same off-set as though it was never saved.

defiante, 2024-02-08:
Factory reset on the machine after the firmware upgrade seems to have fixed the issue for me.  Z-Offset is now saved after running the auto bed level or manually setting the z height

closed 2024-02-19 by CChen616

JozephSchmoe, 2024-07-04:
This issue is not resolved. Factory reset DOES NOT fix this for other users, including me.