QIDITECH/QIDI_PLUS3 #17: file transfers to x-plus3.

WiskyBlackOut, 2024-02-01:
Totally new to this. Cant figure out how to transfer files or find any basic instructions on ways to transfer files to printer

QIDITECH, 2024-02-01:
The guide in the slicing software may help you.
[image]

WiskyBlackOut, 2024-02-01:
> The guide in the slicing software may help you.
> image.png (view on web)

Okay were do I access that? No idea how to get there and searching for
qidislicer doesnt realy result anything.

CChen616, 2024-02-01:
You can download the Qidi Slicer here:
https://qidi3d.com/pages/software-firmware

Additionally, after connecting your printer to Wi-Fi, you can try accessing the printer's Fluidd page from your PC or your mobile phone that connected to the same Wi-Fi network. Simply open a browser and go to:
```
http://{printer's IP address}:10088
```

If you're completely new to 3D printing, you might need to spend some time exploring the Fluidd interface to become familiar with it.

Happy printing!

closed 2024-02-02 by QIDITECH

WiskyBlackOut, 2024-02-13:
Any resources to learn and reference how to use the qidislicer program.