QIDITECH/QIDI_PLUS3 #18: Fluidd Gcode Viewer

ADRS55, 2024-02-01:
G code preview in fluidd does not show the entire print area, part of it is missing.

CChen616, 2024-02-01:
Are you referring to the thumbnail of the model? If so, could you please let us know which slicing software you're using and also send us a copy of the G-code?

ADRS55, 2024-02-01:
No, not thumbnails. I'm referring to the fluidd's gcode viewer, layer progress 2D visualizer.
[image]
[image]

CChen616, 2024-02-02:
Due to Fluidd using the toolhead's minimum and maximum movement ranges to determine the display range for G-code previews, there may be discrepancies with the actual size of the heatbed and what is shown in the slicing software.

As we aim to provide the majority of users with an authentic Fluidd experience, we might not modify this aspect.

ADRS55, 2024-02-02:
I didn't change anything in printer.cfg. 
Information returned from url:

        "axis_minimum": [
          -11,
          -24,
          -6,
          0
        ],
        "stalls": 0,
        "axis_maximum": [
          280,
          280,
          280,
          0
        ],

From my understanding, it should be correct.
But I think it's a bug in fluidd, because today I found this:
[https://github.com/fluidd-core/fluidd/pull/731](url)
It should be fixed in newer versions. Do you plan to update fluidd to a newer version in future releases?
​

CChen616, 2024-02-03:
Thank you for providing a solution!

Currently, it's challenging to update fluidd via update packages. However, you can manually update fluidd by referring to the instructions in the "fluidd" section at:

https://docs.fluidd.xyz/updates/manual

In our machines, fluidd is located at 
```
/home/mks/fluidd/
```
Please do not update other components such as Klipper and Moonraker, as this will cause the printer to malfunction.

ADRS55, 2024-02-03:
I did a manual update as you recommended and the fluidd gcode preview is now working correctly!
[image]

Thank you for your support!

closed 2024-02-03 by ADRS55

dleffler, 2024-03-21:
As a note, if you manually update fluidd, you'll receive warnings on the fluidd interface and warning about unsupported firmware when you ssh in.