QIDITECH/QIDI_PLUS3 #19: Filament replace button behavior

ADRS55, 2024-02-01:
Would it be possible to change the behavior of the filament replace button to the behavior of the filament unload button as it was in previous versions of the firmware? Or add it back? Unloading using the arrow keys is not ideal.

CChen616, 2024-02-01:
As mentioned in our update notes, the old method of material change involved advancing the material slightly before retracting it directly, but this approach poses a risk of clogging the extruder. We understand that many people prefer the old method, but for the sake of machine maintainability, the development team has decided to adopt the new method and, for the time being, not to reintroduce the old button, as it is not recommended.

Of course, as you are currently doing, you can simulate the old method of material change by using the direction keys, but please be aware of the risk of clogging.

ADRS55, 2024-02-01:
I understand, thanks for the explanation.

closed 2024-02-01 by ADRS55