QIDITECH/QIDI_PLUS3 #20: Add Spoolman Support

Senorgif2, 2024-02-10:
I'd like to install an app for Moonraker called Spoolman: https://github.com/Donkie/Spoolman 

This app tracks filament usage, however when attempting to install using SSH I cannot because it requires python 3.9 or newer, and Debian 10 only has 3.7 installed. installing Debian 11 or newer would solve this problem but I assume this would cause other issues. 
The other option is to build and install python 3.9 for Debian 10. I am also unsure if this will cause more issues.

I consider this app essential and part of the reason I purchased this printer was the touted Moonraker software, in which Spoolman is officially supported. If I can't find a way to make this work, I will be returning the printer.

QIDITECH, 2024-02-12:
Sorry, we can't help you, you'll have to upgrade python or any other program you need, but this can cause problems.

leumasnnorht, 2024-02-12:
I do also think that app is essential. I tried to install Spoolman on a separate Server with docker compose, but this also doesn't work, because of moonraker showing an error when I add:

`[spoolman]
server: http://192.168.xx.xxx:7912
#   URL to the Spoolman instance. This parameter must be provided.
sync_rate: 5
#   The interval, in seconds, between sync requests with the
#   Spoolman server.  The default is 5.`

to moonraker.conf

I also tried to add:

`[gcode_macro SET_ACTIVE_SPOOL]
gcode:
  {% if params.ID %}
    {% set id = params.ID|int %}
    {action_call_remote_method(
       "spoolman_set_active_spool",
       spool_id=id
    )}
  {% else %}
    {action_respond_info("Parameter 'ID' is required")}
  {% endif %}

[gcode_macro CLEAR_ACTIVE_SPOOL]
gcode:
  {action_call_remote_method(
    "spoolman_set_active_spool",
    spool_id=None
  )}`

to printer.cfg

But I'am also a bit inexperienced with Klipper, Moonraker and Fluidd. I might be missing something.

Greetings Sam

Senorgif2, 2024-02-12:
Yes I also tried that, and it didn't work. I spoke the developer of Spoolman, and this will only work if Moonraker is updated to a version where it starts supporting Spoolman, The version using the X-plus is older than that. So Even if you updated python to install Spoolman, you also need to update Moonraker to communicate with it. Of course doing any of this yourself will void your warranty. Someday Qidi might update their custom firmware/software, but until then it simply will not work.

pganns, 2024-02-12:
It seems that any version update of fluidd or moonraker is limited to the developers' interest, within the limitations of the MKS motherboard, with Armbian frozen and without updates.

pganns, 2024-02-12:
Someone may need to act as a jailbreaker.

https://github.com/redrathnure/armbian-mkspi

lorangthierry, 2024-02-22:
i have updated moonraker with git pull for the plugin, and my spoolman is installed as Docker on my Unraid. 
in Moonraker.conf i added the ip an port from the Spoolman Docker. 
Everything works fine. 

Changed also from fluidd to mainsail without any problems.

rgnd971, 2024-03-18:
Hello, I can't get Spoolman to run on my nas with OpenMediaVault and Docker.
Can I change the directory that is failing me: spoolman.env "WARNING Data directory is not writable, trying to fix it...
chown: /home/app/.local/share/spoolman: Operation not permitted chown: /home/app/.local/share/spoolman: Operation not permitted".
The /app folder doesn't exist on debian!
If you have any ideas...
And I confirm that Moonraker must be updated, but the directories used will be different.
Thanks in advance!

Translated with DeepL.com (free version)

leadustin, 2024-03-25:
https://github.com/leadustin/QIDI-up2date-english

Phil1988, 2025-01-24:
Hey @Senorgif2 @Samt563 @pganns @lorangthierry @rgnd971 

You can use Spoolman easily with [**FreeDi**](https://github.com/Phil1988/FreeDi).

If you installed an open armbian already based on my **FreeQIDI** guide or if your followed leadustins **QIDI-up2date-english** guide already, you can switch over in like 5min.

Samt563, 2025-01-26:
Thank you for the hint.

I have already installed it and it is much better than the original Firmware, with recent Klipper. The Update Process, when installed, is also much smoother than originally.

Phil1988, 2025-01-26:
Thanks for this feedback, I really appreciate it and happy users like you keeping me motivated :)