QIDITECH/QIDI_PLUS3 #21: Can't update firmware after flashing xindi-20230309.img on a new 32GB EMMC

lucymorn, 2024-03-02:
I had tons of problems with my printer so I purchased the upgraded EMMC and flashed xindi-20230309.img to it using rufus, just as the wiki says to do.
Everything was fine and the printer boots as normal - but the SOC version is 2.0.5 (while UI is 4.2.13)
Connecting the stock white USB device with the latest `QD_Update` folder in its root directory does nothing (meaning the "updating" button doesn't appear)
Placing `QD_Plus_SOC` in the root directory and naming it `mksclient.recovery` brings back the updating button which says to turn off for 20 seconds
Doing that seems to update the printer to the latest firmware version - but then, every time the printer is turned on - it starts copying\updating\writing the screen file of size 20350516 again and again
When it finishes, it boots the printer as normal and the printer is updated and works - but the next time it is turned off and on again - the white screen is back for around 20 minutes to update the screen of size 20350516

What do I do?

CChen616, 2024-03-02:
It seems like you still have your USB device connected to your printer with recovery files in it. Please check your USB device and ensure all recovery files like `mksclient.recovery` or `screen.recovery` are removed.

lucymorn, 2024-03-02:
> It seems like you still have your USB device connected to your printer with recovery files in it. Please check your USB device and ensure all recovery files like `mksclient.recovery` or `screen.recovery` are removed.

That's not the case, of-course. After the update is done I turned off the machine, waited around 30 seconds and disconnected the USB device. Waited some more, then turned it on. Back to a white screen writing size of 20350516

I just did it from scratch, exactly as follow:

1. Flash `xindi-20230309.img.xz` using rufus on one of the EMMCs (either the stock 8GB or the new 32GB)
2. Boot printer with the newly flashed EMMC 
2.1 SoC: 2.0.6 - 
2.2 middle one: 2.10.0 
2.3 UI: 4.2.13
3. Plugged stock USB device with latest `QD_Update` folder in its root directory
3.1 No update button
3.2 Placing  **either** `mksclient.recovery` or `screen.recovery` in the root directory of the USB device will bring back the updating button
4. Update is in progress
5. Update is done, SOC is 4.2.13
6. Turn off the machine and unplug the USB
7. Turn on the machine, white screen, writing 20350516 back again
8. Back to 3

CChen616, 2024-03-02:
Try ssh into your printer and check if any screen update files like `800_480.tft` is in the `/root` directory. Otherwise please provide a list of all the files in the root directory.

lucymorn, 2024-03-02:
> Try ssh into your printer and check if any screen update files like `800_480.tft` is in the `/root` directory. If not please provide a list of all the files in the root directory.

I decided to follow the video in the google drive exactly as is, and only placed 3 files inside the `QD_Update` (removed `800_480_PLUS.tft`) and suddenly the updating button appeared, without needing any of the recovery files. I updated as normal. But the white screen comes back as before. Here is the content of my `/root` directory:

```
root@mkspi:~# pwd
/root
root@mkspi:~# ls
10-armbian-header     gpio-monitor.sh  rk3328-roc-cc.dtb     www
800_480.tft.bak       hid-flash        STM32_HID_Bootloader  xindi
Adaptive_Mesh.cfg     klipper.bin      sysctl.conf	     xinwangka
config.mksini	      merge.py	       TESTGPIO
dev_info.txt	      new_config.ini   uart
gpio-monitor.service  old_config.ini   udp_server
root@mkspi:~# 
```

CChen616, 2024-03-02:
Under normal circumstances, the appearance of a white screen is due to the detection of the `800_480.tft` file in the `/root` folder, which automatically triggers the screen firmware to be updated. Once the update is complete, the file will be renamed to `800_480.tft.bak`.

 During the update check, any `QD_Plus3_UI5.0` file located in the QD_Update directory on a USB device will be renamed to `800_480.tft` and moved to the root directory for the update. Meanwhile, the `800_480_PLUS.tft` file is used for updating the screen via an SD card. Based on the file list you provided, there doesn't seem to be anything unusual. Please restart the machine while ensuring there are no UI update files present to see if the issue still persists.

Thank you for your patience, we had never encoutered this issue before, so it could take some time to troubleshooting.

lucymorn, 2024-03-02:
> Under normal circumstances, the appearance of a white screen is due to the detection of the `800_480.tft` file in the `/root` folder, which automatically triggers the screen firmware to be updated. [...]

Can you please share the part of the code that goes to the screen update?

Also any chance you can upload an updated image to flash the EMMC? My printer isn’t useable in the meantime :/

Problem is easily reproducible - just flash the image and attempt to update the firmware 

any idea why the UI stays up to date despite reflashing of the EMMC?

CChen616, 2024-03-02:
The SOC version is read from a version file stored on the eMMC, while the UI version is obtained directly from the screen. Since you only replaced the eMMC and used the original screen, the UI version will not be affected.

If the problem still exists, try to log in via SSH after booting up and enter the command:
```
journalctl -u makerbase-client.service
```
to view the system service logs, which include detection and updates for the tft file. For example, on my printer, the log:
```
Mar 02 14:18:03 mkspi bash[1393]: 没有找到tft升级文件
```
means that no tft upgrade file was found.

Regarding the new image and related code, I am currently unable to provide them.

lucymorn, 2024-03-02:
> The SOC version is read from a version file stored on the eMMC, while the UI version is obtained directly from the screen. Since you only replaced the eMMC and used the original screen, the UI [...]

I got the printer inside my house from my garage as I got tired of going back and forth and did some more testing  -
* fresh EMMC without firmware upgrade, problem is gone, and I have the following line:
  * `Dec 10 01:34:54 mkspi bash[1477]: 没有找到tft升级文件`
* after a firmware upgrade, the problem comes back, and I have the following line
  * `Dec 10 01:54:53 mkspi bash[1429]: 找到tft升级文件` 

But I can't file any tft file anywhere, like I showed earlier

lucymorn, 2024-03-03:
removing the `QD_Plus3_UI5.0` file from the update seemed to work - but cause a lot of other problems. I am starting to think this image is just broken

CChen616, 2024-03-04:
Yes, if the USB device you are using for updates contains `QD_factory_mode.txt` along with the update files, it may lead to repeated updates occurring. You can check the current version number through the system information interface. If it has already been updated to the latest version, there is no need to update again.

If the issue has been resolved, I will close this issue.

closed 2024-03-06 by CChen616