QIDITECH/QIDI_PLUS3 #22: No wifi \ aic8800 drivers after flashing xindi-20230309.img

lucymorn, 2024-03-03:
I had tons of problems with my printer so I purchased the upgraded EMMC and flashed `xindi-20230309.img` to it using rufus, just as the wiki says to do.
Wifi drivers are no where to be found. 
Trying to get them online and compile them results in the following error -

```
root@mkspi:~/aic8800/drivers/aic8800# make
make -C /lib/modules/5.16.20-rockchip64/build M=/root/aic8800/drivers/aic8800 ARCH=x86_64 CROSS_COMPILE= modules
make[1]: *** /lib/modules/5.16.20-rockchip64/build: No such file or directory.  Stop.
make: *** [Makefile:55: modules] Error 2
```

CChen616, 2024-03-04:
The wifi driver is included in SOC package, so if you have updated to the newest SOC version successfully, it should be able work.

If you want to continue compiling by yourself, here is some suggestions:

1. Our hardware is based on the Rockchip ARM architecture, so you need to change `ARCH=x86_64` to `ARCH=arm` or `ARCH=arm64`.

2. If this does not resolve your issue, it may be due to missing kernel headers or an incorrect link. Try installing the headers with:

```
sudo apt-get update
sudo apt-get install linux-headers-$(uname -r)
```
Then check if the link is correct by running:
```
ls -l /lib/modules/$(uname -r)/build
```
If it doesn't exist, run:
```
# Assuming the kernel headers are located at /usr/src/linux-headers-$(uname -r)
sudo ln -s /usr/src/linux-headers-$(uname -r) /lib/modules/$(uname -r)/build
```
Then try to compile again.

lucymorn, 2024-03-06:
> The wifi driver is included in SOC package, so if you have updated to the newest SOC version successfully, it should be able work.
> 
> If you want to continue compiling by yourself, here is some suggestions:
> 
> 1. Our hardware is based on the Rockchip ARM architecture, so you need to change `ARCH=x86_64` to `ARCH=arm` or `ARCH=arm64`.
> 2. If this does not resolve your issue, it may be due to missing kernel headers or an incorrect link. Try installing the headers with:
> 
> ```
> sudo apt-get update
> sudo apt-get install linux-headers-$(uname -r)
> ```
> 
> Then check if the link is correct by running:
> 
> ```
> ls -l /lib/modules/$(uname -r)/build
> ```
> 
> If it doesn't exist, run:
> 
> ```
> sudo ln -s /usr/src/linux-headers-$(uname -r) /lib/modules/$(uname -r)/build
> ```
> 
> Then try to compile again.

Well, for some reason it doesn't work -
When the Wifi dongle is plugged in it starts in storage mode, even after I switch it to wifi mode using `usb_modeswitch` it still doesn't work as it seems its driver isn't loaded

CChen616, 2024-03-06:
Sry not sure about what storage and wifi mode referring to. But you can toggle the printer between ethernet and wlan by clicking the icon on the right side of the ip address bar, as below:

**ethernet**
[image]

**wlan**
[image]

Also please confirm the wifi dongle model. If it looks like below:
[image]
[image]

It is the newer model and its driver and installation has been included in SOC. If possible, please attach the SOC package you used here for further troubleshooting. 

Or you can try to download the newest SOC package (for some reason, we have updated the V4.2.13 SOC package few weeks ago) and update the printer.

After installation, you can use following command for testing
```
ip link
modprobe aic8800_fdrv
lsmod | grep aic8800
```
and successful example will be:
```
root@mkspi:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether d2:68:61:5c:e3:fb brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DORMANT group default qlen 1000
    link/ether 50:2b:73:94:44:e2 brd ff:ff:ff:ff:ff:ff
root@mkspi:~# modprobe aic8800_fdrv
root@mkspi:~# lsmod | grep aic8800
aic8800_fdrv          417792  0
aic_load_fw            53248  1 aic8800_fdrv
cfg80211              876544  2 8188gu,aic8800_fdrv
```

If the wifi dongle is the older model, it doesn't need driver to be functional.

CChen616, 2024-03-08:
Login as root and move both wifi_driver.gz and driver_install.log to /root dir.
Then run
```
tar -xf wifi_driver.gz
bash driver_install.log
```
Then use 
```
lsmod | grep aic8800
```
to check if wifi mod has been successfully installed.
If run into any error, please paste the error info here.

[driver_install.log](https://github.com/QIDITECH/QIDI_PLUS3/files/14531923/driver_install.log)
[wifi_driver.gz](https://github.com/QIDITECH/QIDI_PLUS3/files/14531924/wifi_driver.gz)

closed 2024-03-22 by CChen616