QIDITECH/QIDI_PLUS3 #23: Amend bed mesh levelling to wait for a couple of minutes after getting the heated bed up to temperature

HostileThuggery, 2024-03-23:
This will allow the build plate to fully equalize in temperature, I've noticed as soon as the bed gets to the target temperature, the bed mesh starts getting taken & the bed can still be warping and warming up in certain areas as this is being done, espeically on a larger build surface such as on my PLUS 3.

Giving this a bit of a 'buffer' or a waiting period before taking the bed mesh would mean everything has equalized before taking any measurements

CChen616, 2024-03-25:
At the current stage, it is recommended to preheat the bed before starting a print to prevent uneven heating. We are considering introducing an option for heating wait time on the printer's screen in future versions.

Minimal-Change, 2024-05-03:
this is what I use for heat soaking for the Q1, for larger sizes I would suggest 10 minutes for 350x350
add these to the Machine start G-Code

M190 S[first_layer_bed_temperature]
G4 P360000; wait 6 minutes for the bed temp to stabilize prior to bedmesh

hope this is able to address your query.