QIDITECH/QIDI_PLUS3 #24: HTTP 500 error

wingman399, 2024-03-29:
Octoprint can connect with the printer. I am able to control everything through fluidd but cannot upload to the printer through my network connection. I get an HTTP 500 error message. I can print through the usb. I updated to the latest release of firmware but it did not fix the issue. I'm not sure where to go from here.

Error uploading to print host:
HTTP 500: {"error": {"code": 500, "message": "Internal Server Error", "traceback": "Traceback (most
recent call last):\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 563, in
move\n os.rename(src, real_dst)\n\nOSError: [Errno 18] Invalid cross-device link:
'/tmp/moonraker.upload-1217.mru' ->
'/home/mks/gcode_files/L1_-_L4_PETG_7h8m.gcode'\n\n\nDuring handling of the above
exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File
\"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 673, in
_process_uploaded_file\n upload_info['dest_path'])\n\n File
\"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 577, in move\n copy_function(src,
real_dst)\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 263, in copy2\n 
copyfile(src, dst, follow_symlinks=follow_symlinks)\n\n File
\"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 122, in copyfile\n copyfileobj(fsrc,
fdst)\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 82, in copyfileobj\n 
fdst.write(buf)\n\nOSError: [Errno 28] No space left on device\n\n\nDuring handling of the above
exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File
\"/home/mks/moonraker/moonraker/app.py\", line 879, in post\n result = await
self.file_manager.finalize_upload(form_args)\n\n File
\"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 532, in
finalize_upload\n result = await self._finish_gcode_upload(upload_info)\n\n File
\"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 604, in
_finish_gcode_upload\n finfo = await self._process_uploaded_file(upload_info)\n\n File
\"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 678, in
_process_uploaded_file\n raise self.server.error(\"Unable to save file\", 500)\n\nutils.ServerError:
Unable to save file\n\n\nDuring handling of the above exception, another exception
occurred:\n\n\nTraceback (most recent call last):\n\n File
\"/home/mks/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1704, in
_execute\n result = await result\n\n File \"/home/mks/moonraker/moonraker/app.py\", line
882, in post\n e.status_code, str(e))\n\ntornado.web.HTTPError: HTTP 500: Internal Server
Error (Unable to save file)\n"}}

CChen616, 2024-03-30:
`OSError: [Errno 28] No space left on device` 

Generally, this issue arises due to the accumulation of too many uploaded gcode files, leading to insufficient disk space. It's recommended to back up and clean up these gcode files or delete old log files.

wingman399, 2024-04-04:
thank you this worked.

closed 2024-04-04 by wingman399