QIDITECH/QIDI_PLUS3 #31: Wifi Password-length is artifically limited

DanielH1987, 2024-05-20:
With WPA2 the maximum amount of characters is 63. The Plus3-Firmware limits this unnecessarily to 32.
Until this is fixed I cannot connect the Plus3 zu my Wifi - because as of now I refuse to change the password on all my IoT-Devices...

Also: support for USB-Keyboard for entering the password would be nice. :)

CChen616, 2024-05-20:
We will fix this issue in the next version. In the meantime, please use a wired connection as a workaround.

Using a USB keyboard instead of screen input is currently a bit challenging (due to the implementation method of the screen), but we will consider this as a potential future improvement.

DanielH1987, 2024-05-20:
I took a look at the code and I see what you mean about the USB-Keyboard.
Other suggestion:
Make it possible to put the Wifi-Credentials on a Text-File on USB-Storage and read from there. That looks simpler.

CChen616, 2024-05-20:
If you can log into the system using a cable, there is a file located at `/etc/wpa_supplicant/wpa_supplicant-wlan0.conf` that stores the Wi-Fi SSID and password. The content of the file looks like this:
```
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
update_config=1

network={
	ssid="test"
	psk="00000000"
}
```

If you know how to create a Deb package, you can make one that only contains this file, and rename it to `QD_Plus_SOC` to make the system think it is an update package (But be care of the package name in control file otherwise is will overwrite the package with same name). This will allow the package to be installed and replace the existing file.

pkefi, 2024-05-31:
Possibility of setting password with keyboard would also make it easier with special characters. I had to create a guest network for printer, because it's on screen keyboard doesn't have all special characters I need.

CChen616, 2024-06-01:
Given the constraints of communication between the screen and the mainboard, it is difficult to transfer USB keyboard input to the screen.

We have already added as many characters as possible to the keypad. Currently, the suggestion is to modify the `/etc/wpa_supplicant/wpa_supplicant-wlan0.conf` file after connecting via an Ethernet cable, or to create a guest Wi-Fi network for the printer.

Another consideration is to leave a backdoor as Daniel said, such as checking for a `QD_wifi.json` file on a USB device at startup. If the file is found, the printer could automatically populate the configuration file with the contents of the `QD_wifi.json` file, which might look like this:
```json
{
  "ssid": "...",
  "pwk": "..."
}
```

DanielH1987, 2024-06-01:
Connecting over ethernet and editing the wpa_supplicant worked fine. Not sure if I have to do this every time I update the firmware or not.
USB keyboard - yeah that seems a bit too much.
But as stated earlier: a JSON or similar on the root of the USB drive with the credentials would be much smoother.

Of course: if you have the camera reading the QR code from your android device would be also a cool option. But that seems to be really out of scope. 😁