QIDITECH/QIDI_PLUS3 #32: Z offset still is not saving

JozephSchmoe, 2024-07-04:
Running "Set Z-offset" or "Auto Bed Leveling" on the X-Plus 3 does not save. Factory reset twice. Please tell us where in the config to store this value. I avoided this firmware for months due to this known, unresolved issue. As it is unresolved, what is the workaround?

CChen616, 2024-07-04:
For the Max3 Plus3 series machines, the Z offset value is stored in the config.mksini file under the babystep section. 
[image]

When the machine detects a connection to Klipper, it reads the value from the file and uses the command `SET_GCODE_OFFSET Z=-{z offset} MOVE=0` to apply the Z offset value to Klipper's runtime variable, gcode offset. You can see this on the Fluidd interface.

[image]

When Klipper restarts, the variable is reset. Generally, the machine detects Klipper reconnecting and sends the command. However, if Moonraker has been restarted, it might cause the system to lose the connection to Klipper, and the command won't be sent when Klipper starts up. Power off and on the printer can get the connection back.

If there's an error in the mksini file, it can also lead to a failure in reading the Z offset. A proper configuration file looks like this:

[image]

Factory reset will reset the `config.mksini` file.

The above explanation assumes operations are performed using only the screen. To avoid conflicts, the Klipper file has been modified so that after running `PROBE_CALIBRATE`, the saved offset value in `printer.cfg` will be zero.

We will work to optimize this logic and include updates in the next release.

dleffler, 2025-03-21:
Though there are few more values in the 4.2.15 firmware to help with the app, I'm having the same problem.  babystep is correct based on the last Set Z Offset, but it is NOT set after powering back on.