QIDITECH/QIDI_PLUS3 #33: XPlus 3 is abandoned?

Bonnyfede00, 2024-07-10:
Is there any hope that the quidi xplus 3 printer will still be updated? When will klipper and moonraker be updated to support the obico commands?

CChen616, 2024-07-11:
There are still updates related to QIDILink support and security etc, but they do not include updates for `Klipper` or `Moonraker`.

noisymime, 2024-07-25:
I think Qidi needs to make some decisions about how the 3 series devices are being supported and marketed. 

We all know that the firmware is now significantly out of date with very down level versions of the packages that are most important to the user (Klipper, Moonraker etc) as well as a core OS (Armbian Buster) that contains multiple known security issues. 

Despite all of this, these printers are still listed under the 'New Printers Release' on the Qidi website. This seems misleading at best and dishonest at worst. Either these are new, supported and updated printers or they're considered legacy, it can't be both. It would be greatly appreciated by the community if there was some more information made available about Qidi's intentions around these devices. 

(None of this is should be seen as a criticism, in particular of @CChen616 who has provided amazing support and communication on technical matters, but the silence from an official standpoint is deafening now).

QIDITECH, 2024-07-26:
There are many reasons why a package or system cannot be updated, including hardware limitations, long update times, and the probability that the update will fail. For these reasons, plus3's system and software cannot be upgraded to the latest version, but we have not given up optimizing its features.
Q1 Pro has added an adaptation to QIDI Link, but this has led to some network security issues that we have been working on recently. When this feature is stable, we will extend it to other models.
(And @CChen616  is a member of our staff, and we have been doing our best to address users' questions.)

Psychotica85, 2024-08-07:
Sorry Qiditech but this is waste. I have a Q1 Pro and a XPlus3 device. The XPlus3 is much better in print performance as the Q1 Pro. The Q1 doestn satisfied me. Bed meshing is bad and the nozzle cleaning too. The Filamentholder ist poor of quality and the software is old as in the XPlus3. 

We dont need a QIDI Link - we need actual software and drivers. A device from 2024 (Q1) with microcontroller firmware and software from 2022. Good joke. Other customer could update the software to the actual version (look at different gits) and its work, but QIDI cant? Mainly Qidi printers are good devices but something goes wrong with QIDI about the Software, the Shopdescriptions and the whole descriptions of your additional items on the sites.

NightHammer1000, 2025-01-02:
I have to agree with this. The Terrible Software Support on the XPlus 3, and the fact that I had to heavily mod it myself to get a modern klipper version to run made sure i will NEVER buy a QIDI Product ever again.

Bonnyfede00, 2025-01-04:
They could make all the "codes" public (to enable the use of features like the screen, for example) so that the community can keep this printer alive even after its official "lifecycle" ends. I noticed that the hardware of the Xplus 4 seems very similar to that of previous models, so I don't understand why Klipper and other software cannot be updated to the latest version. Let’s hope the developers have a change of heart and consider this...

Samt563, 2025-02-14:
Hello, I would simply use https://github.com/Phil1988/FreeDi 

for me it just works! :)

Greetings Sam