QIDITECH/QIDI_PLUS3 #36: x-plus 3 doesnt boot after update

dsmal150, 2024-08-08:
Hello, i have problem with x-plus 3 printer after update klipper via web interface, printer wrote: MCU protocol error, older version firmware runing on MCU. I try update via USB flash disk, but printer now wrote: System start abnormaly. Please what i can do try now?  Thanks for any advice

CChen616, 2024-08-09:
It seems that you accidentally upgraded Klipper. The firmware update only replaces some of the Klipper files and doesn't fully restore Klipper to the normal version. You can try installing this update package, which will only replace the core Klipper files with the correct version. After installation, try installing firmware version 4.2.13 again.
[QD_Plus_SOC.zip](https://github.com/user-attachments/files/16554757/QD_Plus_SOC.zip)

dsmal150, 2024-08-09:
Thanks, at end i replaced klipper and moonraker folder from github and reinstall firmware via usb. And most importantly, i shut off notification for new versions. :)  Printer is now fully functionaly. Can i asked if you planing upgrade klipper etc, for this printer some in future?

leadustin, 2024-08-19:
> Can i asked if you planing upgrade klipper etc, for this printer some in future?

In my opinion, this will never happen as it would mean too much effort for Qidi. The risk of users with less experience breaking the printer would be very high. If you want to update the printer at your own risk, you can use my repository or CCheng616´s 
https://github.com/leadustin/QIDI-up2date-english
https://github.com/CChen616

astur35, 2024-11-14:
> Gracias, al final reemplazó klipper y moonraker carpeta de github y reinstalar firmware vía usb. Y lo más importante, aleje la notificación para nuevas versiones. :) La impresora es ahora totalmente funcional. Puedo preguntar si usted planchando klipper de actualización, etc, para esta impresora algunos en el futuro?

Could you tell me what steps you followed to achieve it. The same thing happens to me as you but I can't get it to boot with the USB placed on the card directly. And right now I have a big paperweight.

dsmal150, 2024-11-14:
I connected with xplus via ssh with help mobaxterm. I deleted folder klipper from /home/mks/. Maybe is better only rename klipper folder. And delete, after successful repair. Klipper i download from https://github.com/QIDITECH/klipper and rename to klipper.  If you have updated moonraker then moonraker too. And after that shutdown printer, for a while. And after recover update oficial qidi firmware.