QIDITECH/QIDI_PLUS3 #40: New Xplus3 update issue

baron011182, 2024-10-17:
I have updated the printer today using the new release build for this device and I have encountered an issue.  

When I updated it I got a bank screen mentioning Data Error.  I managed to recover it but now I'm getting the following error on the screen and when http into the printer.  

Include file '/home/mks/klipper_config/MKS_THR.cfg' does not exist

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

This is on the new build that Qidi released 15hrs ago.  How do I fix this issue.

baron011182, 2024-10-17:
[image]

TrollPlayer94, 2024-10-18:
How did you manage to recover from the DATA ERROR because nothing is working for me?

svalsorim, 2024-10-19:
I just have white screen after the latest update, i think I just got message update failed check message. What message? Where? Also looking for a way to recover it but i have no working screen anymore

baron011182, 2024-10-19:
I managed to get the screen working by using the QD_Plus3_UI5.0 to mksscreen.recovery and putting that file onto a usb stick.  Enter the usb stick in the printer usb port and switch on.  It will do a re-flash on the screen to get that working.

TrollPlayer94, 2024-10-19:
I got too this screen now to after trying this fix, but don't know how to proceed further.

baron011182, 2024-10-19:
I think this has corrupted the klipper and moon raker folder under /home/mks/

Once I ssh into it using mobxterm and copied the files over from qidi own version of Klipper and moon raker I was able to get working again.

svalsorim, 2024-10-19:
Great! Thank you for the update ill try that!

svalsorim, 2024-10-20:
Mine will not accept mksscreen.recovery no matter what i do. I could access it via network but now I lost that too

QIDITECH, 2024-10-21:
According to your description, I think it is caused by your updating of klipper or moonraker. Restoring it to the default version of firmware should solve the problem.

baron011182, 2024-10-21:
Considering I’ve not updated klipper or moonraker.  Only your firmware.  This update is what caused me the issue.  Sent from my iPhone

Plexix, 2024-10-21:
I don't understand how Qidi could release such a firmware version like 4.2.15. Not only does it cause the screen to freeze with a white screen of death, but if a user manages to recover using a micro SD card, there's a risk of damaging the printer. While the procedure with the micro SD card may resolve the white screen issue, it doesn't fix the underlying problems with the firmware itself. After switching to version 4.2.15 and performing the initial calibration, like input shaping and setting the Z offset, the printer completely ignores these settings during printing. If you start a print with mesh bed leveling, the printer immediately begins to dig the nozzle into the bed during the initial test line. First, I dealt with the frozen white screen, and now I've damaged my print bed. Qidi can't be serious about this.

baron011182, 2024-10-21:
I had this issue too.  Qidi is not taking this one serious.  Always the end user fault which is not the case here. 

> On 21 Oct 2024, at 14:19, Plexix ***@***.***> wrote:
> 
> nozzle

svalsorim, 2024-10-21:
I was going to comment that none of us updated klipper but used official update, but I left it to OP to address it. I’m still dealing with white screen and “data error! “ message.

whb0514, 2024-10-22:
> ![9973D552-7E12-4DC3-A7B9-1408BED9710E](https://private-user-images.githubusercontent.com/98152266/377579467-5dd0c438-542b-4fc7-a007-8652fc7b413d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk1NTgwNDYsIm5iZiI6MTcyOTU1Nzc0NiwicGF0aCI6Ii85ODE1MjI2Ni8zNzc1Nzk0NjctNWRkMGM0MzgtNTQyYi00ZmM3LWEwMDctODY1MmZjN2I0MTNkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyMlQwMDQyMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTMzN2ZiNmFkYWRkY2JmYzk4ZDZlMDcyYzBjM2U2OGZlNjMwNWQxZmE2MzllYjM1MzI1OTIwY2QwYTUyOTUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cKrlt_puIaHR5ch3fk8oIzVzw17oPc6NMG_MOSf1Dt8)

Just need the parameter page in the fluidd page, create a MKS_THR.cfg file, save it, and restart the machine to solve this problem. As for why the file is missing, we will look for the reason.

baron011182, 2024-10-22:
Qidi have now gone quiet 🤐.  Can we have some support and at least pull this firmware from your firmware page.  This needs to be looked at and recompiled for the xplus 3.Sent from my iPhone

herby99999, 2024-10-22:
> > ![9973D552-7E12-4DC3-A7B9-1408BED9710E](https://private-user-images.githubusercontent.com/98152266/377579467-5dd0c438-542b-4fc7-a007-8652fc7b413d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk1NTgwNDYsIm5iZiI6MTcyOTU1Nzc0NiwicGF0aCI6Ii85ODE1MjI2Ni8zNzc1Nzk0NjctNWRkMGM0MzgtNTQyYi00ZmM3LWEwMDctODY1MmZjN2I0MTNkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyMlQwMDQyMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTMzN2ZiNmFkYWRkY2JmYzk4ZDZlMDcyYzBjM2U2OGZlNjMwNWQxZmE2MzllYjM1MzI1OTIwY2QwYTUyOTUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cKrlt_puIaHR5ch3fk8oIzVzw17oPc6NMG_MOSf1Dt8)
> 
> Um dieses Problem zu lösen, benötigen Sie lediglich die Parameterseite auf der Fluidd-Seite, erstellen eine MKS_THR.cfg-Datei, speichern sie und starten den Computer neu. Wir werden nach dem Grund suchen, warum die Datei fehlt.

kann den link nicht öffnen wäre dankbar damit der weiße screen weg geht

gptech2444, 2024-10-23:
Wish I read this before doing the update. This is ridiculous. Now I have to try and unbrick my printer.

mane81, 2024-10-25:
I am experiencing the same issue. Please remove this firmware release and correct it.

baron011182, 2024-10-25:
Qidi are now totally ignoring this thread.  Sent from my iPhone

logiclinux, 2024-10-25:
I have just updatet to .Ver 4.2.15 got a blank screen but acces to the Printer via SSH. Then i found the solution to download the file 800_480_PLUS.tft on an sdcard and put this card into the cardslot at the Qidi X-Plus 3 screen (very complicated to reach i had to unscem the screen to rech the slot)  Swiching off wait 30 sec. swich on and it installs the frimware for the Screen.
But now i have no more access to the Network adjust,ments via this screen but you see the softkey gets pressed (at the moment no need caus its connected very well via ethernet and wifi). So i have SOC Ver 4.2.15 amd UI Ver 4.2.13  MMC is mot diplayed . When the srenn goes isn sctreensaver mode i cant wake him up by touching it.

svalsorim, 2024-10-25:
For the record, I had not the same problem as the OP, I just had white screen( UI part failed) I got it resolved by repeated mksscreen.recovery updated in my case 5th consecutive time update passed. Maybe try forcing the update few times to see will it resolve your issue. There is a write up in wiki page for a Qidi q1 pro what to do if update failed.

JozephSchmoe, 2024-11-03:
I was able to get past the white screen issues only to run into the Z offset problems. In my case, using the screen to set the Z offset results in a ridiculous -2.68 value. Adding 2mm to that value, for -.68 got me pretty close to what the value should be. This is far from ideal but I can at least print. You'll want to edit the config.mksini file, for anyone who didn't learn that from the last disastrous firmware update.
Do a reboot and Fluuid should see the correct value.
[image]