QIDITECH/QIDI_PLUS3 #42: update error

Lovelace3314, 2024-10-28:
update fw from V4.2.13 to V4.2.15.I found that I could not complete the platform calibration,I resolved the problem by modifying the configuration file.

[image]

[image]

[image]

Lovelace3314, 2024-10-28:
[gcode_macro M4029]
gcode:
    BED_MESH_CLEAR
    G91 
    G1 Z10 F600
    G90
    G28
    G1 X{printer.toolhead.axis_maximum.x/2 - printer.probe["x_offset"]} Y{printer.toolhead.axis_maximum.y/2 - printer.probe["y_offset"]} F7800
    G1 Z15 F600
    ABORT
    PROBE_CALIBRATE
    ABORT
    G0 Z0

Lovelace3314, 2024-10-29:
This error exists in the file I downloaded on October 18th, and the problem has been fixed in the file I downloaded today.

[image]

[image]

[image]

Lovelace3314, 2024-10-29:
The date of revision of the two documents does not coincide.
Good thing you've solved the problem.
I used an old upgrade package to cause the above problem.

[image]

QIDITECH, 2024-10-30:
We replaced the wrong version as soon as we uploaded it. Your action was too fast.

closed 2024-10-30 by Lovelace3314