QIDITECH/QIDI_PLUS3 #43: Z offset in latest firmware

JozephSchmoe, 2024-11-03:
Klipper doesn't seem to be pulling the z offset value from the config.mksini file. Nozzle digs deep into print bed every time. I have factory reset but so far, nothing is fixing it. Editing the printer.cfg file does nothing, of course, due to the code that resets the value to 0 in the printer.cfg. At this point I can't use my printer. Is Qidi aware of this?

[image]
[image]

QIDITECH, 2024-11-04:
Please try to edit z_offset in saved_variables.cfg.
[image]

dleffler, 2025-03-21:
Where is the saved_variables.cfg file located?

Olokoft, 2025-05-02:
Up , i have same problem , i don't have saved variable file..