QIDITECH/QIDI_PLUS3 #45: 4.2.15 Detected Virus

WMU-JBLAINE, 2024-11-04:
Tried downloading this firmware update and it's being blocked due to a detected virus.

QIDITECH, 2024-11-05:
We use frp for some data communication. Because it has the function of transmitting data, it is always identified as a virus by security software. But rest assured, he doesn't have any malicious features.
You can learn more by following the links below:
https://github.com/fatedier/frp/issues/2095
If you still feel that it is not secure, please do not update the firmware.

jjl40, 2024-11-16:
So do you think it may be helpful to explain how to download it? I've tried it with Chrome and Edge. Cannot download.
If you really want to be helpful, you could move into the 21 century and improve the update process. Instead of being so condescending.

WMU-JBLAINE, 2024-11-16:
Given the issues reported by other users I would not install this update anyways.  Multiple issues reported.  And the only way to download the update is either  to tell AntiVirus to ignore the issue, shut off AntiVirus, or use a system with no AntiVirus.  Yeah not very secure or user friendly.  It's espcially interesting since none of the previous Firmware releases had frp that caused this issue.

jjl40, 2024-11-16:
Thank you for the prompt reply.

Yes, I have done several updates since receiving my Plus 3 over a year ago and never had an issue. It was still a pain in the neck.
I’m wondering if Qidi is leaving me behind or just trying to get the new products ironed out.

Cheers

John Lytle

QIDITECH, 2024-11-18:
This update only adds network communication and ADAPTS to QIDI Link. Other features have not changed, if you are concerned about this warning, please do not update the firmware.

gfxdsnr427, 2025-01-04:
Based on QIDITECH replies and all the issues I'm having, this will be my first and last QIDI Tech printer.   I'm trying to update the firmware because all the issues I'm having. Half the time the screen tells me to TURN OFF and TRY AGAIN LATER when booting up. When it does boot, most times the screen goes blank when trying to read the USB. Yeah, I have to use the USB drive, because even though the printer can see the network, it won't connect. Once the stars align and it finally gets to the print screen and I press the print button, sometimes it just sits there not warming up, all temps 0. If I cancel, it NEVER finishes the process to stop the print and I have to hard-reset, which most of the time brings me back to the TRY AGAIN LATER screen. The printer prints great when I can get it to print, but this ain't worth the hassle.  If you're looking to get any QIDI Tech printer, I would say look elsewhere.