QIDITECH/QIDI_PLUS3 #46: Replacing TMC2209 with TMC2240

leadustin, 2024-12-25:
Hello,

is it possible to replace the original TMC2209 with TMC2240? Are the pins the same? I know that I have to make changes to the printer.cfg. I just want to know if there is compatibility with the mainboard.
Thx

QIDITECH, 2024-12-26:
The motor drive is integrated on the motherboard and cannot be replaced.

closed 2024-12-26 by QIDITECH

leadustin, 2024-12-27:
> The motor drive is integrated on the motherboard and cannot be replaced.

So the drivers are soldered on directly? What would be the process for a defective driver? Buying a new mainboard is out of the question. Can the defective driver then be desoldered and replaced with a 2240?

QIDITECH, 2024-12-28:
Sorry, the number of pins in the two chips is different, so this is not possible.